innoveit/react-native-ble-manager #358: Do not call `byteArrayToWritableMap` with a null byte array

Sean-Brown, 2018-07-17:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.4.0
- react-native v0.52.3
- iOS/Android v7.1.1

### Expected behaviour

My log should not constantly get these errors:
```
01:57 W/System.err: java.lang.NullPointerException: Attempt to get length of null array
01:57 W/System.err: 	at android.util.Base64.encode(Base64.java:494)
01:57 W/System.err: 	at android.util.Base64.encodeToString(Base64.java:456)
01:57 W/System.err: 	at it.innove.Peripheral.byteArrayToWritableMap(Peripheral.java:257)
01:57 W/System.err: 	at it.innove.Peripheral.asWritableMap(Peripheral.java:158)
01:57 W/System.err: 	at it.innove.BleManager.getConnectedPeripherals(BleManager.java:480)
01:57 W/System.err: 	at java.lang.reflect.Method.invoke(Native Method)
01:57 W/System.err: 	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:374)
01:57 W/System.err: 	at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
01:57 W/System.err: 	at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
01:57 W/System.err: 	at android.os.Handler.handleCallback(Handler.java:751)
01:57 W/System.err: 	at android.os.Handler.dispatchMessage(Handler.java:95)
01:57 W/System.err: 	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
01:57 W/System.err: 	at android.os.Looper.loop(Looper.java:154)
01:57 W/System.err: 	at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
01:57 W/System.err: 	at java.lang.Thread.run(Thread.java:762)
```

### Actual behaviour

I am constantly getting errors about trying to encode a null byte array. I believe the issue is in [Peripheral.java, line 174](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java#L174) -- it should be checking the result of `entry.getValue()` and only call `byteArrayToWritableMap` if the value is non-null.

marcosinigaglia, 2018-07-24:
Hi, try to use the latest release.

Sean-Brown, 2018-07-24:
Oops didn't mean to close but I'll check it out later and close the issue if it appears to be resolved. Thank you!

Sean-Brown, 2018-07-24:
Yes I'm still seeing the error:
```
01:55 W/System.err: java.lang.NullPointerException: Attempt to get length of null array
01:55 W/System.err: 	at android.util.Base64.encode(Base64.java:494)
01:55 W/System.err: 	at android.util.Base64.encodeToString(Base64.java:456)
01:55 W/System.err: 	at it.innove.Peripheral.byteArrayToWritableMap(Peripheral.java:264)
01:55 W/System.err: 	at it.innove.Peripheral.asWritableMap(Peripheral.java:164)
01:55 W/System.err: 	at it.innove.BleManager.getConnectedPeripherals(BleManager.java:480)
01:55 W/System.err: 	at java.lang.reflect.Method.invoke(Native Method)
01:55 W/System.err: 	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:374)
01:55 W/System.err: 	at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
01:55 W/System.err: 	at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
01:55 W/System.err: 	at android.os.Handler.handleCallback(Handler.java:751)
01:55 W/System.err: 	at android.os.Handler.dispatchMessage(Handler.java:95)
01:55 W/System.err: 	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
01:55 W/System.err: 	at android.os.Looper.loop(Looper.java:154)
01:55 W/System.err: 	at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
01:55 W/System.err: 	at java.lang.Thread.run(Thread.java:762)
```

Looking at the code I can't imagine how the [`getConnectedPeripherals` method in the BleManager.java file](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/BleManager.java#L473) could work. It constructs a new peripheral object but calls the constructor that won't initialize the `advertisingDataBytes` attribute. Then it calls the `asWritableMap` method on the new peripheral, which will call [`byteArrayToWritableMap` with the advertisingDataBytes member as an argument](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java#L164), but I don't see how this member will ever be non-null. I'm not sure what the correct solution would be but a couple come to mind:

- Initialize advertisingDataBytes to an empty array
- Do a null-check on advertisingDataBytes before calling byteArrayToWritableMap

What I don't fully understand is how this would impact the caller (BleManager.getConnectedPeripherals)

Sean-Brown, 2018-07-24:
I experimented with initializing the advertisingDataBytes to an empty array and it seems to avoid the error, i.e. on [line 48](https://github.com/innoveit/react-native-ble-manager/blob/master/android/src/main/java/it/innove/Peripheral.java#L48) do `private byte[] advertisingDataBytes = new byte[0]`

closed 2019-06-19 by marcosinigaglia