innoveit/react-native-ble-manager #362: Use of undeclared identifier 'RCTBridgeWillReloadNotification' in iOS

Symous, 2018-07-23:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.4.0
- react-native v0.45.1
- iOS/Android iOS 11.4

### Expected behaviour

running normally

### Actual behaviour

Xcode give me an error that Use of undeclared identifier 'RCTBridgeWillReloadNotification'

### Reason

the RCTBridge of react-native 0.45.1 do not contains the RCTBridgeWillReloadNotification

marcosinigaglia, 2018-07-24:
Hi @Symous , your RN version is old, can you try to search for the same problem in other project?

Symous, 2018-07-27:
@marcosinigaglia , yes, I understand. but the readme said that this lib support RN0.40 or higher which may misleading users like me. :)

closed 2019-06-18 by marcosinigaglia