innoveit/react-native-ble-manager #364: the BleManager.startNotification is error

yydinfo, 2018-07-28:
when you use 
 BleManager.startNotification(this.state.id, this.state.uuidData.nofityServiceUUID[0], this.state.uuidData.nofityCharacteristicUUID[0])
            .then(() => {
                console.log('Notification started');
                this._write()
            })
            .catch((error) => {
                console.log('Notification error:',error);
            });

and then the log is '''Could not find service with UUID 0000FFE0-0000-1000-8000-00805F9B34FB on peripheral with UUID 1910A25A-94B4-4F9D-A08E-83DD6EDC55C7''' , why?

Rana-Muhammad-Amir, 2018-08-10:
Facing same error but in error.....
any help please....

 'Notification error', 'Characteristic 13333333-3333-3333-3333-333333330003 not found'

eddroid, 2018-08-11:
Did you call `retrieveServices` first?

_Before write, read or start notification you need to call retrieveServices method_
https://github.com/innoveit/react-native-ble-manager#note

closed 2018-08-20 by marcosinigaglia