innoveit/react-native-ble-manager #366: retrieveServices never returns after disabling and enabling ble from device settings

anwarhamr, 2018-08-02:
### 
    "react-native-ble-manager": "^6.5.0",
    "react": "16.3.1",
    "react-native": "0.55.4",
    Android for now, will verify iOS soon (but I don't think it's a problem)

### Expected behaviour
Get the services like it does the first time when my app starts and is connected.

### Actual behaviour
hangs and never returns.

### Steps to reproduce

1. Start android app using lib
2. connect to a device and retrieveServices; works great.
3. leave app running and go to ble settings on phone turn off and back on
4. scan for devices
5. device is found again via scan
6. call connect(); connects just fine
7. call retrieveServices() it never returns...

```
  connect(peripheral) {
    this.log.warn('BLE Connect reqeusted', peripheral);
      return BleManager.connect(peripheral)
      .then(()=>{
        this.log.warn('BLE Connected to', peripheral);
        return BleManager.retrieveServices(peripheral) //<-- never returns after first success.
          .then(peripheralInfo=>{
            this.log.warn('BLE Service info:' ,peripheralInfo);
            return Promise.resolve(peripheralInfo);
          });});
     }
```

JJSrra, 2018-09-12:
Having a similar issue in iOS, reproducing the same steps, but instead of step 8 it throws a "Peripheral not found or not connected" error.

@anwarhamr @marcosinigaglia Any news on that?

anwarhamr, 2018-09-13:
@JJSrra I have not had a chance to validate w/ my iOS device.  I'll try to get to it this week.

JJSrra, 2018-10-01:
@marcosinigaglia Please could you take a look at that? Seems like a bug and is quite annoying as the library works fine until it gets to this point... Thanks in advance!

@anwarhamr Did you test it on your iOS device? Any news? 🙂

anwarhamr, 2018-10-08:
@JJSrra This is not happening on my ios devices.

JJSrra, 2018-10-09:
I'm also rescanning the same way you do @anwarhamr (as reported in the issue) but it doesn't seem to work. First connection when I start the app is always successful, and even first reconnection after disconnecting is successful most of the times, but then second reconnection hardly ever works, it just throws the "Peripheral not found or not connected" even though the app still connects to the device. Given the fact that I'm able to reconnect once, I assume that my code has a good structure and there has to be something else.

Related to your issue with Android, have you tried using the `refreshCache` method? It's an Android-only method but perhaps is related to this since it refreshes the peripheral's services and characteristics.

nagaokashi, 2018-10-19:
#391

nagaokashi, 2018-10-19:
>  then second reconnection hardly ever works, it just throws the "Peripheral not found or not connected" even though the app still connects to the device. 

Did you check  `getConnectedPeripherals ` at the second reconnection? If no, I think you should try.

JJSrra, 2019-01-22:
Hi everyone, sorry for the late response.

@nagaokashi I tried to connect and then check the `getConnectedPeripherals` value and it always returns the expected value (the device I'm looking for), even at the second reconnection in which `retrieveServices` fails. As I said, it works fine the very first time, and even the first time I try to reconnect. The library is able to connect to the device and retrieve the services, but sadly the second time hardly ever works ☹️

The above happens if I create a new `BleManager` instance each time I try to connect, calling `BleManager.start()`. If I disconnect from a device and then try to connect to the same device several times (without "restarting" BleManager) it works, but it fails if I try to connect to more than one BleManager instance, i.e. when I try to connect to several devices in the same app. Switching between devices works once or twice, but the next time I get the same wrong behaviour after calling `retrieveServices`.

@marcosinigaglia Any updates on this? Could you work on it on past months?

anwarhamr, 2020-10-03:
I have not seen this issue in a long time.  My pattern is to always call `getConnectedPeripherals` check that list for my client, if it's not there I call `connect`.  This is working for me.

closed 2020-10-03 by anwarhamr