innoveit/react-native-ble-manager #368: Re connecting issue

HarshalValanda, 2018-08-08:
### Version

Tell us which versions you are using:

-  react-native-ble-manager: v6.2.9
- react-native v0.55.3"
- iOS v11.2.6

### Expected behaviour

First time and second time connection successfully -> RetriveServices -> StartNotification -> Get data
after that i disconnect and reconnect peripheral but retriveServices 

Tried to RetriveServices but after 2 or 3 connection can't get services

connection and other stuff working 2 or 3 times after that can't get proper response 

my question: 
How can i connect and. disconnect peripheral?
is there any number of times restriction in iOS?

JJSrra, 2018-10-09:
This is the same I'm experiencing, see also #366 @marcosinigaglia.

Did you find a way to go through this?

marcosinigaglia, 2018-10-09:
Hi, I had no time to check this. I hope to have some spare time in the next weeks.

NN01010101, 2018-10-26:
The same thing happens to me. After a lot of testing, I have observed, that "disconnect" does not really work. I have peripheral, that stops advertising once a connection with the smartphone is established. If I restart the app while being connected, the android Bluetooth keeps maintaining the connection to the peripheral. Before my app attempts to connect again, it checks with getConnectedPeripherals for connected devices. And indeed – I get the list of connected peripheral id(s), and if I use connect(id) the connection is established, but if I try disconnect(id) I get "peripheral not found" error. Since I can not discover the peripheral, I'm confident that the phone still maintaining the connection to the peripheral. When I restart Bluetooth, everything works fine, but only until the next "disconnect" that I assume does not work.
I also tried to refreshCache before disconnecting from the peripheral – no effect or sometimes I get "Attempt to invoke virtual method \'java.lang.Class java.lang.Object.getClass()\' on a null object reference".

JJSrra, 2019-01-22:
I'm experiencing the same as @HarshalValanda using iOS, but perhaps you should also look into #366 @NN01010101, in case that helps you in any way. I just updated with my latest testing and `retrieveServices` is not working yet for me.

@marcosinigaglia Could you take a look at this? Thanks in advance!

JJSrra, 2019-01-22:
Sorry, could you specify the code snippet that I need to add? I'm not sure if by "start function" you mean the `init` function, and also pasting `[self init]` throws an error when compiling with Xcode, probably because I'm not doing it properly.

JJSrra, 2019-01-22:
Nevermind, I found the `start` function, sorry about that 😅

That worked!!! And I realized it's not in the latest `BleManager.m` file in the library. I think you should make a Pull Request for @marcosinigaglia to add it in the next release. Thanks a lot!

JJSrra, 2019-01-22:
@HarshalValanda In case you don't want to submit a Pull Request for any reason I can make it myself, but I prefer you to do it since you came up with the solution 🙂

pkyeck, 2019-01-25:
fix worked for me - can someone with more knowledge check if this is actually the right thing to do and update the module? thanks!

pkyeck, 2019-01-25:
@marcosinigaglia I had the same problem that first connection worked perfectly but after disconnecting and reconnecting the `BleManager.retrieveServices(peripheral.id)` threw an error saying `Peripheral not found or not connected`.
I didn't change any JS code, just added the `[self init]` to the `start` method and reconnecting is working as expected.

pseudocode:

```
import BleManager from 'react-native-ble-manager';

class XX extends React.Component {
  constructor(props) {
    super(props);

    const BleManagerModule = NativeModules.BleManager;
    this.bleManagerEmitter = new NativeEventEmitter(BleManagerModule);
  }

  componentDidMount() {
    // add all this.bleManagerEmitter.addListener ...
    // finally start
    BleManager.start({ showAlert: false });
    BleManager.checkState();
  }

  async componentWillUnmount() {
    await BleManager.disconnect(selectedPeripheral.id);
    // remove all this.bleManagerEmitter.addListener
  }

  onBTChange(args: any) {
    if (args.state === 'on') {
      BleManager.scan([], 30, true);
    }
  }
  
  onDiscoverPeripheral = async (peripheral) => {
      try {
        if (peripheral.advertising.localName === 'xxx') {
          await BleManager.stopScan();

          // connect
          await BleManager.connect(peripheral.id);

          // THIS FAILS AFTER RECONNECTING TO THE SAME DEVICE
          await BleManager.retrieveServices(peripheral.id);

          await BleManager.startNotification(...)
        }
      } catch (err) {}
  };
}

marcosinigaglia, 2019-01-25:
So the problem appear after you put the app in background? The start is called only in componentDidMount.

pkyeck, 2019-01-25:
ah sorry, should have mentioned that I do all this on an overlay -> so:

1) show overlay -> (constructor) -> componentDidMount -> connect ...
2) hide overlay -> componentWillUnmount -> disconnect ...
3) show overlay -> componentDidMount -> connect ... => ERROR

pdbq21, 2019-04-04:
> @marcosinigaglia I had the same problem that first connection worked perfectly but after disconnecting and reconnecting the `BleManager.retrieveServices(peripheral.id)` threw an error saying `Peripheral not found or not connected`.
> 
> I didn't change any JS code, just added the `[self init]` to the `start` method and reconnecting is working as expected.
> 
> see: https://github.com/pkyeck/react-native-ble-manager/blob/master/ios/BleManager.m#L283
> 
> pseudocode:
> 
> ```js
> import BleManager from 'react-native-ble-manager';
> 
> class XX extends React.Component {
>   constructor(props) {
>     super(props);
> 
>     const BleManagerModule = NativeModules.BleManager;
>     this.bleManagerEmitter = new NativeEventEmitter(BleManagerModule);
>   }
> 
>   componentDidMount() {
>     // add all this.bleManagerEmitter.addListener ...
>     // finally start
>     BleManager.start({ showAlert: false });
>     BleManager.checkState();
>   }
> 
>   async componentWillUnmount() {
>     await BleManager.disconnect(selectedPeripheral.id);
>     // remove all this.bleManagerEmitter.addListener
>   }
> 
>   onBTChange(args: any) {
>     if (args.state === 'on') {
>       BleManager.scan([], 30, true);
>     }
>   }
>   
>   onDiscoverPeripheral = async (peripheral) => {
>     try {
>       if (peripheral.advertising.localName === 'xxx') {
>         await BleManager.stopScan();
> 
>         // connect
>         await BleManager.connect(peripheral.id);
> 
>         // THIS FAILS AFTER RECONNECTING TO THE SAME DEVICE
>         await BleManager.retrieveServices(peripheral.id);
> 
>         await BleManager.startNotification(...)
>       }
>     } catch (err) {}
>   };
> }
> ```

I have the same problem

pdbq21, 2019-04-04:
> ah sorry, should have mentioned that I do all this on an overlay -> so:
> 
> 1. show overlay -> (constructor) -> componentDidMount -> connect ...
> 2. hide overlay -> componentWillUnmount -> disconnect ...
> 3. show overlay -> componentDidMount -> connect ... => ERROR

I have the same problem

puneetkumarC, 2019-05-02:
I had the same problem and it is resolved thank you 
kudos 
@HarshalValanda @pkyeck

cmcleese, 2019-05-10:
Im trying the suggested solution with the `[self init];` added to the **start** function.
However xcode keeps telling me that `expression result unused`.
I tried this proposed fix on a couple different versions of the library but always the same result and always same issue.

Moreover, the recent change in v6.6.4 [here](https://github.com/innoveit/react-native-ble-manager/pull/490/commits/bb6e3fe9331fd6ccb0b926cc99ffef23baa0c928)
doesn't help either.
This version and the previous release version both do not show the device upon "re-scan" after the device was turned off and on again.

blendimaliqi, 2022-01-21:
> Im trying the suggested solution with the `[self init];` added to the **start** function. However xcode keeps telling me that `expression result unused`. I tried this proposed fix on a couple [...]

@cmcleese  Did you find anything out about the 'expression result unused' message? Im having the same issue.

closed 2023-02-10 by marcosinigaglia

rakshitbharat, 2025-01-09:
its 2025 and this issue is still available atleast for me

astout, 2025-02-11:
If anyone else is encountering reconnect issues on Android, I have found that you should also use `removeBond(deviceId)` in addition to `disconnect(deviceId)`. This fixed the issue for me. i.e. I monitor the BLE Manager for errors connecting with or communicating with the device and if so, we go through a sequence of disconnect/reconnect.