innoveit/react-native-ble-manager #370: .Start after the first call does not request to the user to turn off BT Antenna

anwarhamr, 2018-08-13:
### Version
6.5.0
Tell us which versions you are using:

- react-native-ble-manager v6.5.0
- react-native v55.4
- iOS only works first call; Android works fine

### Expected behaviour
When `start` is called a check should always be made to determine if the antenna is on.
### Actual behaviour
Subsiquent calls to `start` succeed but do not ask the user to turn on BT

### Steps to reproduce

1. Start app with BT off
2. call .`start` user is prompted to go turn it off via Settings
3. turn BT on via settings
4. scan devices found
5. got back to settings and turn of BT
6. scan or call `start` do not ask the user to turn on BT again.

marcosinigaglia, 2018-08-20:
The start function is to init the module so you have to call this function only one time. You have to check manually after that if the BT is on.

anwarhamr, 2018-09-20:
@marcosinigaglia so I have to keep track of when it's turned off then inform the user manually to turn it on?  Is there not a way to request that iOS turn it on similar to what is done on start?

closed 2019-03-07 by anwarhamr