innoveit/react-native-ble-manager #374: Question: Would this support reading car data from OBD2 Modules?

kyle-ssg, 2018-08-23:
This is more of a question so I appreciate if it is out of scope.

I've got a react native app using this module connecting to an OBD2 Vehicle device, do you know if it would be feasible to use this library to display vehicle sensors like standard obd2 apps, e.g. engine rpm?

marcosinigaglia, 2018-08-23:
I don't know how the OBD2 device works but if you have to read data from a ble device this library is ok.

kyle-ssg, 2018-08-24:
I'm closing this as it definitely is possible by finding the write service/characteristics to write to then sending something like
1. connect to device
2. find the service/characteristic you need to subscribe to in order to send commands
3. send (base64 encoded) AT SP 0

If anyone finds this and wants to learn how, ping me a message.

kyle-ssg, 2018-08-24:
PS Thanks for the great work

closed 2018-08-24 by kyle-ssg