innoveit/react-native-ble-manager #376: Android can not receive updated BLE advertisement data

stevesuyao, 2018-08-29:
### Version
- react-native-ble-manager v6.5.0
- react-native v0.51.0
- Android v.7.0

### Expected behavior

I have a BLE peripheral which is broadcasting real-time data. The Android mobile should scan to get the peripheral continuously changed advertisement data 

### Actual behavior
My Android could get the ADV data, but will not update it.  Only after I stop the scan and start the scan again, the ADV data would be updated.

stevesuyao, 2018-08-29:
### I found a Solution
change 
https://github.com/innoveit/react-native-ble-manager/blob/fbd0d13197692f0d86a0426a38e7ae89f635d2a8/android/src/main/java/it/innove/LollipopScanManager.java#L111

to`perpheral.updateData(result.getScanRecord().getBytes());` could resolve the issue.

Not sure if it is a bug.

CatapultJesse, 2018-12-12:
Can confirm this fix works.  Made a PR: https://github.com/innoveit/react-native-ble-manager/pull/420

closed 2018-12-27 by marcosinigaglia