innoveit/react-native-ble-manager #378: Write mehtod returns nothing

szkkeny, 2018-09-13:
### Version

- react-native-ble-manager v6.5.0
- react-native v0.56.0
- iOS v.11

### Expected behaviour
We can get the value which is sent by peripheral with Write mehtod.

### Actual behaviour
there is no return value.

marcosinigaglia, 2018-09-13:
Hi, is not an error, the write method doesn't return value.

anwarhamr, 2018-09-20:
You will have to read the value after writing it if you need confirmation or set notification on (if possible) on the characteristic.

harshit275, 2019-01-23:
@anwarhamr Can you please elaborate some more on the read, write and set notification of this library as to how to use them. I am trying to fetch the data from a PID of a BLE using write method of this library but I am getting empty promise object. Don't know what I am doing wrong.

closed 2019-02-01 by marcosinigaglia