innoveit/react-native-ble-manager #379: ble disconnect abnormal

shengliZh, 2018-09-13:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.0
- react-native v0.56
- iOS/Android  iOS 11.4/Android 6.0 or above

### Expected behaviour

Expected behaviour: after ble connect, notification and write cannot use so long time and should no disconnect so many times.

### Actual behaviour

notification use as long as 8s, and write use 2s. Some times before complete notification, ble Peripheral did disconnect, but I don't know why it has disconnect, this happens often.

### Steps to reproduce

1.connect success
2.bond device
3.retrieveServices
4.notification
5.write

### Stack trace and console log
First Time:
2018-09-13T12:55:45     connected device CC:E1:5D:1A:70:14  10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:55:46     1526 notify success: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:55:46     1528 notify success: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:55:46     1530 notify success: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:55:46     send authCode success: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A

Second Time:
2018-09-13T12:56:17     connected device CC:E1:5D:1A:70:14  10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:56:51     1526 notify failed: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A ,error: Peripheral did disconnect: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A

2018-09-13T12:56:51     disconnect device: 10639B5B-2B26-B9AD-DFF0-5898BD68F17A SIREN_L

2018-09-13T12:56:52     scan device will delay 1s

marcosinigaglia, 2018-09-13:
Hi, are you sure is not the ble device that disconnect?

srikanthkyatham, 2018-09-13:
For me I get this error 
ReactNativeBleManager: onConnectionStateChange to 0 on peripheral: CB:BF:89:5E:35:DA with status: 8 bondState: 10

Sequence of steps
1. startNotifications
2. series of read callbacks
3. disconnect due to -> ReactNativeBleManager: onConnectionStateChange to 0 on peripheral: CB:BF:89:5E:35:DA with status: 8 bondState: 10
resulting in BleManagerDisconnectPeripheral event on the javascript side.

Expected
1. startNotifications
2. series of read callbacks
3. javascript to initiate disconnect

The vary same steps works on iOS with the same device

shengliZh, 2018-09-14:
Hi, @marcosinigaglia 
After the joint debugging, I found that basically every time I enable notification, I will encounter problems.
At this time I have meet below error:
Failed to set client characteristic notification for 0000XXXX-XXXX-XXXX-XXXX-785feabcd123

shengliZh, 2018-09-14:
Hi, @srikanthkyatham 
I think you have get ble connect timeout, this will happens in android, such as status:8、status:19, and some times I have get status:133