innoveit/react-native-ble-manager #382: ios fatal exception

shengliZh, 2018-09-20:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.0
- react-native v0.56
- iOS/Android v.11.4 iOS/v6.0 Android

### Expected behaviour

Ble will connect and run normal
### Actual behaviour

RCTFatalException: Exception '*** Collection <__NSSetM: 0x282417a40> was mutated while being enumerated.' was thrown while invoking write on target BleManager with params ( "A01D2AAD-711B-528B-C698-6B785C71E81C", "00001523-1212-EFDE-1523-785FEABCD123", "00001530-1212-EFDE-1523-785FEABCD123", ( 119, 0, 255, 255, 255, 255 ), 20, 2979 ) callstack: ( 0 CoreFoundation 0x00000001df773f10 <redacted> + 252 1 libobjc.A.dylib 0x00000001de941a40 objc_exception_throw + 56 2 CoreFoundation 0x00000001df773790 <redacted> + 0 3 SirenRN 0x0000000104fa0dd8,

But, I have not had this error in Android. And it high probability  in ios

### Steps to reproduce
only connect one device and I have set scanner 2min
1. connect (also in scanning)
2.notification (also in scanning)
3.write (also in scanning)

srikanthkyatham, 2018-11-07:
It happened to me while invoking read instead of write

marcosinigaglia, 2018-11-07:
Hi, it's a known problem.

closed 2019-02-01 by marcosinigaglia