innoveit/react-native-ble-manager #383: Always disconnect after 20 - 30second after connect peripheral

hidayatNgganteng, 2018-09-25:
### Version
- react-native-ble-manager v6.5.0
- react-native v0.50.4
- iOS v.6s

### Expected behaviour

after I connect, it should not disconnect continuously (abnormal)

### Actual behaviour

after connecting to peripherals, about 20 seconds will disconnect. and this will continue

### Steps to reproduce

1. connect peripheral
2. success
3. disconnect continue

marcosinigaglia, 2018-10-01:
Hi @hidayatNgganteng , probably is the device that disconnect you.

hidayatNgganteng, 2018-10-01:
i am using https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral
for testing, is this problematic? even though I didn't disconnect

Thanks @marcosinigaglia

ShooterArk, 2018-12-26:
Hi, did you find the issue ? I am still facing this one. I can successfully connect to a device but immediately after it gets disconnected. I am not an expert in BLE so I dont know whats the issue here. Here is my log data, 

`12-26 17:27:50.693 22450-22489/? D/ReactNativeBleManager: Connect to: 3E:CB:01:0B:DF:50
     Less than M
     Trying TRANPORT LE with reflection
12-26 17:27:50.694 22450-22489/? W/System.err: java.lang.NoSuchMethodException: connectGatt [class android.content.Context, class java.lang.Boolean, class android.bluetooth.BluetoothGattCallback, class java.lang.Integer]
        at java.lang.Class.getMethod(Class.java:664)
        at java.lang.Class.getDeclaredMethod(Class.java:626)
        at it.innove.Peripheral.connect(Peripheral.java:110)
        at it.innove.BleManager.connect(BleManager.java:228)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
12-26 17:27:50.695 22450-22489/? W/System.err:     at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:363)
        at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:162)
        at com.facebook.react.bridge.queue.NativeRunnable.run(Native Method)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:31)
        at android.os.Looper.loop(Looper.java:135)
12-26 17:27:50.696 22450-22489/? W/System.err:     at com.facebook.react.bridge.queue.MessageQueueThreadImpl$3.run(MessageQueueThreadImpl.java:194)
        at java.lang.Thread.run(Thread.java:818)
12-26 17:27:50.696 22450-22489/? D/ReactNative:  Catch to call normal connection
12-26 17:27:50.697 22450-22489/? D/BluetoothGatt: connect() - device: 3E:CB:01:0B:DF:50, auto: false
    registerApp()
    registerApp() - UUID=0d25d762-3877-440c-885c-f2b7bd0c0cb6
12-26 17:27:50.698 16165-16217/? D/BtGatt.GattService: registerClient() - UUID=0d25d762-3877-440c-885c-f2b7bd0c0cb6
12-26 17:27:50.698 22450-22489/? D/ReactNativeBleManager: onConnectionStateChange to 2 on peripheral: 3E:CB:01:0B:DF:50 with status0
12-26 17:27:50.698 16165-19380/? D/BtGatt.GattService: onClientRegistered() - UUID=0d25d762-3877-440c-885c-f2b7bd0c0cb6, clientIf=5
12-26 17:27:50.698 22450-22469/? D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
12-26 17:27:50.699 22450-22450/? D/BluetoothGatt: discoverServices() - device: 3E:CB:01:0B:DF:50
12-26 17:27:50.699 16165-16215/? D/BtGatt.GattService: clientConnect() - address=3E:CB:01:0B:DF:50, isDirect=true
12-26 17:27:50.701 22450-22489/? D/ReactNativeBleManager: Peripheral event (BleManagerConnectPeripheral):3E:CB:01:0B:DF:50
12-26 17:27:50.702 22450-22489/? D/ReactNativeBleManager: Connected to: 3E:CB:01:0B:DF:50
12-26 17:27:50.708 16165-19419/? D/BtGatt.GattService: discoverServices() - address=3E:CB:01:0B:DF:50, connId=null
12-26 17:27:50.709 16165-19419/? E/BtGatt.GattService: discoverServices() - No connection for 3E:CB:01:0B:DF:50...
12-26 17:27:50.726 22450-22488/? I/ReactNativeJS: Connected to 3E:CB:01:0B:DF:50
12-26 17:27:51.637 22450-22489/? D/ReactNativeBleManager: Retrieve services from: 3E:CB:01:0B:DF:50
12-26 17:27:51.637 22450-22489/? D/BluetoothGatt: discoverServices() - device: 3E:CB:01:0B:DF:50
12-26 17:27:51.638 16165-16215/? D/BtGatt.GattService: discoverServices() - address=3E:CB:01:0B:DF:50, connId=null
12-26 17:27:51.638 16165-16215/? E/BtGatt.GattService: discoverServices() - No connection for 3E:CB:01:0B:DF:50...
12-26 17:27:57.504 16165-19384/? D/HeadsetStateMachine: Disconnected process message: 10, size: 0
12-26 17:27:58.781 548-1755/? W/ActivityManager: getRunningAppProcesses: caller 10084 does not hold REAL_GET_TASKS; limiting output
12-26 17:28:00.531 548-631/? E/WifiStateMachine: WifiStateMachine CMD_START_SCAN source -2 txSuccessRate=0.38 rxSuccessRate=0.28 targetRoamBSSID=any RSSI=-73
    WifiStateMachine starting scan for "PTCL-BB...1"WPA_PSK with 2412`

Please advise. Thanks

Nagavikas96, 2020-06-18:
Same issue... Any help?

headkaze, 2021-02-14:
I faced this issue on Android. You should install nRF Connect and see if you can get a stable connection in that.

What I'm noticing is if I do not select the "Pair & connect" dialog it will disconnect after 30 seconds. So go to your Bluetooth settings and forget the device then attempt to connect and pair again. For some reason the dialog pops up twice and if I don't click "Pair & connect" on both occasions it will disconnect again.

closed 2023-02-10 by marcosinigaglia