innoveit/react-native-ble-manager #390: Getting my own device bluetooth address?

willypraditya, 2018-10-15:
I need to get my Bluetooth Address for a  react-native project I've been working on. I've been searching for the right dependency to use. I was wondering if this package can get my own device bluetooth address.

Thanks

willypraditya, 2018-10-15:
> Hi, you can't with this library and is not suggested to use your ble mac address in real scenario.

Hi there, thanks for replying. Can you tell me the why it's not recommended? I need to get the Bluetooth address for my final project creating a smart home security that connects react-native with a raspberry pi that locks / unlocks the door when the user's bluetooth is in range.

marcosinigaglia, 2018-10-22:
Hi, you need to scan for the peripheral, not to insert manually the address.

closed 2019-02-01 by marcosinigaglia