innoveit/react-native-ble-manager #391: Not work after turned Bluetooth on -> off -> on.

nagaokashi, 2018-10-16:
### Version

Tell us which versions you are using:

react-native-ble-manager v ^6.4.0
react-native v 0.55.4
iOS/Android v 10.3/8.0

### Expected behaviour

App should work normally after turned Bluetooth on -> off -> on.

### Actual behaviour

After turned Bluetooth on -> off -> on, app does not work.
See the picture.
[image]

At the second red line, a notification should started same as the first red line. But everything stop without any console log.
What happened ?

```
```

chentianci123, 2019-11-28:
Hello, what method do you call to turn on Bluetooth?

HKSOLDIER, 2020-10-09:
for turning on bluetooth.BleManager.enableBluetooth()
  .then(() => {
    // Success code
    console.log("The bluetooth is already enabled or the user confirm");
  })
  .catch((error) => {
    // Failure code
    console.log("The user refuse to enable bluetooth");
  });

zeeshan6, 2022-07-17:
> for turning on bluetooth.BleManager.enableBluetooth() .then(() => { // Success code console.log("The bluetooth is already enabled or the user confirm"); }) .catch((error) => { // Failure code console.log("The user refuse to enable bluetooth"); });

Not working on Android 12 app gone carsh

closed 2023-12-21 by marcosinigaglia