innoveit/react-native-ble-manager #393: RetrieveServices not working on Release Builds

nikoWah, 2018-10-18:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.1
- react-native v0.57.2
- Android v.8

### Expected behaviour
Retrieve Services should work in both Debug and Release Builds

### Actual behaviour
I have the correct peripheral ID and the log showed that the Connect method is working. Also, I can see on the device that the app is connected.

**RetrieveServices is not working in Release Builds, even though it's working in Debug Builds.**

Is there any way I can log the errors from RetrieveServices. The Catch error method doesn't seem to work. 

Or did I miss any settings?

Any help is appreciated. Thanks.

marcosinigaglia, 2018-10-18:
Hi @nikoWah , I never tested with an Android 8 so I don't know if there is an issue. Try to debug.

marcosinigaglia, 2018-10-18:
I've never had a problem like this...maybe can be a permission problem.

nikoWah, 2018-10-19:
@marcosinigaglia I found the problem. 

I used Buffer.from to convert the hex string into bytes array for the write command. It seemed like the Buffer library is either missing or linked improperly in React Native. After using a NPM module, it worked.

Thanks for your time.

closed 2018-10-19 by nikoWah