innoveit/react-native-ble-manager #394: startNotification fails

INanaay, 2018-10-18:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.4.0
- react-native v0.55.2
- iOS/Android v7.0

### Expected behaviour

I can connect to my device without any problems. I can retrieve services, read and write to it.

I'd like to use startNotification to set a listener. startNotification should be working,

### Actual behaviour

startNotification fails.

### Stack trace and console log

The log is ''Set notification failed for xxxxx-xxxxx-xxxxx-xxxxx'' 

Does anyone have any idea ? Thanks in advance.

marcosinigaglia, 2018-10-18:
Hi, are you sure that the characteristic support the notification?

INanaay, 2018-10-18:
Yes, we use an app called BLE Scanner to test our bluetooth device. 

What we do is we send messages every second. With that app, we can see Notification as a property, and when we activate the notifications on that app, it reads what we send without any trouble.

marcosinigaglia, 2018-10-18:
Try to add a delay beetween retrieveServices and startNotification.

INanaay, 2018-10-18:
I'll try that, thank you.

INanaay, 2018-10-23:
Hi,

We can't manage to make this work. Even with a timeout, startNotification fails.

This is our code : https://pastebin.com/UXeWhtup

And our log : https://pastebin.com/WeAgv7ex

The id is : const id="30:AE:A4:1B:9F:76"
The serviceID is : const serviceUUID= '80dc4c84-831a-4937-9058-2cf0bf28b8c8'
The characteristicID is : const characteristicUUID='4d8b9541-0159-4b4c-801a-03ec5520bd8a'

marcosinigaglia, 2018-10-23:
Try to check with BleManagerConnectPeripheral if the peripheral is connected because there is an issue in android.

INanaay, 2018-10-23:
I added a listener before calling connecting, and it is indeed triggered.

closed 2019-02-01 by marcosinigaglia

nolife08021, 2021-01-15:
@INanaay  @marcosinigaglia hi, i having the same issue. "Set notification failed for 000036f6-0000-1000-8000-00805f9b34fb"

nolife08021, 2021-01-15:
retrieve result
` [{"characteristic": "36f5", "properties": {"Write": "Write"}, "service": "fee7"}, {"characteristic": "36f6", "properties": {"Notify": "Notify", "Read": "Read"}, "service": "fee7"}, {"characteristic": "fec7", "properties": {"Write": "Write"}, "service": "fee7"}, {"characteristic": "fec8", "properties": {"Indicate": "Indicate", "Read": "Read"}, "service": "fee7"}, {"characteristic": "fec9", "properties": {"Read": "Read"}, "service": "fee7"}]`

logcat
`2021-01-15 15:09:09.673 23627-23864/com.lock2020 D/ReactNativeBleManager: startNotification
2021-01-15 15:09:09.674 23627-23864/com.lock2020 D/ReactNativeBleManager: registerNotify
2021-01-15 15:09:09.674 23627-23864/com.lock2020 D/ReactNativeBleManager: setNotify
2021-01-15 15:09:09.674 23627-23864/com.lock2020 D/BluetoothGatt: setCharacteristicNotification() - uuid: 000036f6-0000-1000-8000-00805f9b34fb enable: true
2021-01-15 15:09:09.679 23627-23863/com.lock2020 I/ReactNativeJS: Set notification failed for 000036f6-0000-1000-8000-00805f9b34fb
`