innoveit/react-native-ble-manager #395: BleManagerStopScan should be invoked on stopScan()

rnevius, 2018-10-22:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.0
- react-native v0.56.0
- Android v9

### Expected behaviour

The `BleManagerStopScan` event should be invoked when `stopScan()` is called.

### Actual behaviour

`BleManagerStopScan` is only called when a scan finishes automatically (i.e. after `seconds` have elapsed since calling `scan()`) or when the scan fails.

### Steps to reproduce

Call `BleManager.stopScan()`. No event is triggered.

smdjeff, 2019-06-11:
was it fixed?

MD3XTER, 2019-06-13:
I still have this issue

rnevius, 2019-06-14:
A fix for a (related?) iOS bug was [committed](https://github.com/innoveit/react-native-ble-manager/commit/837ee9249dd3ee85f0937b9adb25ca07dc2fe4c6) a couple of days ago, but I don't believe this issue is resolved on Android @marcosinigaglia

marcosinigaglia, 2019-06-14:
Yes, Is fixed only on iOS, my fault.

marcosinigaglia, 2019-06-14:
Try the new release.

MD3XTER, 2019-06-14:
> Try the new release.

Is this issue fixed for iOS in this release? `v6.6.4`

marcosinigaglia, 2019-06-14:
In iOS is fixed in v6.6.5.

MD3XTER, 2019-06-14:
But, it's not released yet, right?

MD3XTER, 2019-06-14:
> In iOS is fixed in v6.6.5.

I updated the library and `BleManagerStopScan` is still not triggered. @marcosinigaglia did it work on your end?

p.s I don't think I need to relink dependencies, but just in case, do I need to do something else after updating the library from npm?

marcosinigaglia, 2019-06-18:
@MD3XTER do you still have the problem?

MD3XTER, 2019-06-18:
Yes, latest version, didn’t fix it for me

marcosinigaglia, 2019-06-18:
Can you debug the library code on xcode? Are you sure that you update the library?

MD3XTER, 2019-06-18:
I will check if the latest changes in your release are present in node_modules as well. I will debug in xcode. Should I just watch the log?

marcosinigaglia, 2019-06-18:
Check if you arrive in this line https://github.com/innoveit/react-native-ble-manager/blob/c6189ecf9f81905f750039d22e36691974a2993e/ios/BleManager.m#L357

MD3XTER, 2019-08-28:
@marcosinigaglia I tested now the code and it seems like it works. I must have missed something. I think it's safe to close this issue, thank you for the fix :)

closed 2019-08-29 by marcosinigaglia