innoveit/react-native-ble-manager #397: Can't write other characteristics after startNotification for specific characteristic.

rnevius, 2018-10-26:
This may not actually be a bug, as my understanding of BLE writes is a bit limited.

### Version

- react-native-ble-manager v6.5.0
- react-native v0.56.0
- Android v9

### Expected behaviour

`startNotification()` is called for characteristic `A`. `write()` is subsequently called for characteristic `B`. Characteristic `B` is written successfully.

### Actual behaviour

`startNotification()` is called for characteristic `A`. Writes are blocked for all other characteristics until characteristic `A` notifies of a new value.

### Steps to reproduce
```
await BleManager.retrieveServices(peripheralId);

// Start notification for characteristic "A"
await BleManager.startNotification(peripheralId, serviceUUID, characteristicA_UUID);

// The following doesn't write a value for characteristic "B" and doesn't throw an error.
await BleManager.write(peripheralId, serviceUUID, characteristicB_UUID, data);
```

harshit275, 2019-01-23:
@marcosinigaglia I am facing this same issue, tried debugging it few time don't know what I am doing wrong. Please check my code below:
`setTimeout(() => {
  BleManager.startNotification(peripheral.id, serviceUUID, characteristicUUIDN)
    .then(() => {
      console.log('Started notification on ' + peripheral.id);
      setTimeout(() => {
        debugger;
        var temp = BleManager.write(peripheral.id, serviceUUID, characteristicUUIDW, pids)
          .then(() => {
            console.log('Writed PID: ' + pids);
          }).catch((error) => {
            console.log('Error while writing: ' + error);
          });
        console.log(temp);
      });
    }).catch((error) => {
      console.log('Error while notification: ' + error)
    });
});`
Please help me out with this.

rnevius, 2019-01-26:
@harshit275  Did you call `retrieveServices()` before `startNotification()`? I don't see it in the code you shared.

closed 2019-06-18 by marcosinigaglia

yoonachen, 2021-12-14:
Hey, do you solve this problem?
I have the same issue ...

rnevius, 2021-12-14:
I did not.