innoveit/react-native-ble-manager #401: cannot use refreshBleCache() method

yoshimoto8, 2018-11-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.0
- react-native v55.4
- Android v7.0

### Expected behaviour

refresh remaining cache

### Actual behaviour

Although I tried to refreshCache before disconnecting from the peripheral,  "Attempt to invoke virtual method 'java.lang.Class java.lang.Object.getClass()' on a null object reference" always error occur.

### Steps to reproduce

call 'refreshBleCache(PeripheralId)' before disconnect method. Error occur

### related issues

#368

nagaokashi, 2018-11-08:
Try to use refreshCache  after connected and before retrieveServices.
Note: do not use '.cache'.
```
if (Platform.OS === 'android') {
      	BleManager.refreshCache(peripheral.id)
		  .then((peripheralInfo) => {
		    // Success code
		    console.log('cache refreshed!', peripheralInfo);
		    connectAndPrepare(peripheral, data_wr); 
		  }); 
      }
```

marcosinigaglia, 2019-06-18:
Hi, I don't use this function in my project, is an internal function to force the refresh of the peripheral.

closed 2019-06-18 by marcosinigaglia