innoveit/react-native-ble-manager #403: Reading data from ble device after writing

KayhanB, 2018-11-14:
### Version

- react-native-ble-manager v6.5.1
- react-native v0.57.4
- Android

Hi

Our ble device runing on a custom electronic card . When i writing data to device, card  returning me data. but i can't get that data .We tested it with BLE Scanner  app it is on the appstore and its working perfectly.  BLE Scanner Writing and getting what we should to see.
 I've tried to use read() function after write function  but read function returning me something strange.
How can i do this and  what am i doing wrong

**This is my  console after writing and reading.**

```
Peripheral is connected
Writed :  566666
ReadData:  Ô69ØCT
```

### **This is my code**

```
write(message, device, service, characteristic) {

    let byteMsg = stringToBytes(message);
    BleManager.write(device.peripheral, service, characteristic, byteMsg)
      .then(() => {
        console.log('Writed : ', message)
        BleManager.read(device.peripheral, service, characteristic, byteMsg)
          .then((readData) => {
            this.setState({ connecting: false, connectingId: null, writedId: device.peripheral, writeErrorDeviceId: null }, () => {
              let readedData = bytesToString(readData)
              console.log('ReadData: ', readedData)
              this.disconnect(device.peripheral)
            })
          }).catch((err) => console.log(err))
      }).catch((err) => {
        console.log(err, " Error Write")
        this.setState({ connecting: false, connectingId: null, writedId: null, writeErrorDeviceId: device.peripheral })
        this.disconnect(device.peripheral)

      })

  }

```

marcosinigaglia, 2018-11-14:
Hi, are you sure that you have to read? Maybe you need to use notification.

KayhanB, 2018-11-14:
I have read your documentation but i dont understand what is notification for and im sure after read function return 'Ô69ØCT' that string

marcosinigaglia, 2018-11-16:
What did you expect to read?

KayhanB, 2018-11-17:
ble device runing on our card. This card listening ble device and  When i send  something to ble device   card return me answer through ble device.  İ want to read that answer. When i trying to read device , result is  "Ô69ØCT" wrong .We changing answer on card but result is same every time. This is the my problem but  no need to read any more we solved this problem with another way .
thank you for taking the time and answering

marcosinigaglia, 2018-11-20:
Are you sure that is not an encoding problem? What kind of data are you reading?

MrDockal, 2018-11-21:
Hi @KayhanB in order to get an understanding of the BLE communication, I would recommend you to go through these posts
[Very nice explained here](https://www.silabs.com/community/wireless/bluetooth/knowledge-base.entry.html/2018/05/30/ble_basics_masters-i4n9)
[This medium post is more about programming](https://medium.com/@avigezerit/bluetooth-low-energy-on-android-22bc7310387a)
[Download the starter kit, which contains more theory](https://www.bluetooth.com/develop-with-bluetooth/build/developer-kits/bluetooth-starter-kit)

If you want to check your peripheral profiles, services and characteristics I would recommend you to use (nRF Connect for android)[https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=cs]

Hope it helps you to get the problem solved.

closed 2018-11-21 by KayhanB

KayhanB, 2018-11-21:
@MrDockal thank you for your interest

malikowaisghafoor, 2020-06-13:
Do you have any idea how I can see BLE SCANNER data on run time in any Android device?