innoveit/react-native-ble-manager #412: Advertised Manufacturer Data is Different between iOS and Android

danielwgraf, 2018-11-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v 6.5.1
- react-native v 0.57.4
- iOS/Android v. 12.1/8.0

### Expected behaviour

Peripheral info should be the same between iOS and Android 

### Actual behaviour

Android seems fine and has the desired services/characteristics/data.

iOS doesn't have all of the services/characteristics and the data is different (doesn't contain the name or scan data).

### Steps to reproduce

I connected to the same device with an Android device and an iOS device (with a development build)

marcosinigaglia, 2018-12-04:
Can you debug it? I don't thinks is a library problem.

danielwgraf, 2019-01-02:
So as of right now I've worked around the problem by making platform specific functions to read the data. Both phones had the data I needed, but the byte arrays were different, so I couldn't use the same method to find what I needed.

closed 2019-02-21 by marcosinigaglia