innoveit/react-native-ble-manager #415: Does getConnectedPeripherals detect already connected devices?

jkpvecino, 2018-12-03:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v0.57.5
- Android v.8.1

### Question

I was wondering if getConnectedPeripherals still only works with devices that have been prescanned or if you can retrieve the information of a device that's already connected to the phone. I'm essentially attempting to retrieve the information of a headset that I've connected to my phone.

marcosinigaglia, 2018-12-04:
Probably the headset is paired or bonded, I never see if this kind of devices are listed.

jkpvecino, 2018-12-04:
So if I pair the phone with a pair of headphones or a headset, it will only show up in the array returned by the BondedPeripherals method and not the ConnectedPeripherals method?

closed 2021-02-18 by marcosinigaglia