innoveit/react-native-ble-manager #418: Android scan not returning any device

amiBrd, 2018-12-09:
### Version 

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v ^0.55.4
- Android 8.1

### Expected behavior

I have a loop that: 
- connects to one of many advertising devices 
- does some logic (read/write/notify) 
- disconnects and move to the next device in the list.
- Once every configured time in between iteration, a scan is invoked to identify devices
In case there is a communication error with one device, it is removed from the list (unless, discovered again by a new scan).

This logic should work, even if devices are shut down and restarted, or bluetooth is disabled/enabled, 

### Actual behavior
I don't have steps to reproduce, but I have observed (and it may take a couple of minutes or a couple of hours), that suddenly all devices are dropped from the list after failure to communicate, and from that moment on, scan returns 0 peripherals and I fail to communicate with the devices in an endless loop.
Bluetooth on/off, does not resolve this, only shutting down the application and opening it again resolves this.
When this happens, another app such as light blue (on the same phone, or a different one, can identify all the peripherals correctly).

### Steps to reproduce
NA
Have tried to pinpoint the issue, but was not able to. 

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

ShooterArk, 2018-12-26:
Did you find any solution? I am facing the same issue. Cant scan any device as well. I am using the example App to test the bluetooth mechanism but its facing this issue. 

Please advise.

Sanjeevs1995, 2019-01-28:
Please look into this issue.

this package does not support new devices

NataliaCt22, 2019-02-06:
Turn on location from your cellphone and try again.

Sanjeevs1995, 2019-02-07:
@NataliaCt22  Tried with location ON still no progress

closed 2019-02-21 by marcosinigaglia

DanielPiflaks, 2019-05-07:
Did someone find any solution for this problem? 
I facing the same issue.

Sanjeevs1995, 2019-05-11:
How can one close the issue where a proper solution is not found

pumpkin123, 2019-09-06:
I  faced the same problem.
Some device need LOCATION authority.

Solution: Open LOCATION authority in the device.