innoveit/react-native-ble-manager #423: didn't work at ios

Handby, 2018-12-25:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?   6.5.4
- react-native v?.?   0.57.8
- iOS/Android v.?   iOS12.1.2

install success but did't work.
even i turn off ble, nothing happend.

tal987, 2018-12-25:
@Handby same, I have trying on simulator and on physical device.. when I do search for devices I can't find anything and cant connect to any device even tho Im giving Mac address its not connect

marcosinigaglia, 2018-12-26:
You can't use the ble on a simulator, you need a device.

tal987, 2018-12-26:
@marcosinigaglia I have been trying that on physical device which connect to my laptop. I mean I run the local version(debug) on physical device and can’t find any device

srikanthkyatham, 2019-01-17:
Would be helpful to paste some code. And logs.

closed 2019-02-01 by marcosinigaglia