innoveit/react-native-ble-manager #424: Cant find any device with iOS

tal987, 2018-12-25:
Hey guys I have been trying to discover device and connect but non of that work. I wouldn't be able to find any device and I know there are a few devices that are advertising.

my code is:

       <componentDidMount() {
      BleManager.start({showAlert: true}).then(() => {
      // Success code
      console.log('Module initialized');
    });
  }>

<startScan() {
      BleManager.scan([], 5, true).then((results) => {
        console.log('Scanning...'
        );
      });

      BleManager.connect('B8211E4A-0989-6282-9970-5A1B2764A921')
  .then(() => {
    // Success code
    console.log('Connected');
  })
  .catch((error) => {
    console.log('NOTConnected');

    // Failure code
    console.log(error);
  });
  
  BleManager.getConnectedPeripherals([]).then((results) => {
    if (results.length == 0) {
      console.log('No connected peripherals')
    }
    console.log(results);
  });
    
  }>

nothing comes up... what am I doing wrong here?

Thanks

marcosinigaglia, 2019-01-11:
Hi @tal987 ,are you handle the BleManagerDiscoverPeripheral event?

tal987, 2019-01-11:
@marcosingaglia what do you mean by handle? The code I added in my post is all I do except to the render. But I can’t even connect or discover any device. Im trying that with physical device that the develop app version installed on the device

srikanthkyatham, 2019-01-15:
@tal987 could you answer the following 

1. which platform are you using iOS or Android. 
2. You using it in foreground I assume Yes?
3. are you trying to connect to ibeacon or anything else ?

tal987, 2019-02-01:
@srikanthkyatham 1. IOS
2. Yes
3. First im trying only to scan and find something... I check with my iPhone for any bluetooth connection around me and there's a few: my Mac, my AirPods and more but can't find anything...

closed 2019-02-06 by tal987

Ilario17, 2019-02-28:
@tal987 have you found a solution?

srikanthkyatham, 2019-02-28:
For me I was not able to scan on the background, apple insists on having the service uuids in the scan. Otherwise in the foreground you should be able to find all the beacons. I did not have any issue using the module. 

Assuming You have followed setup instructions in setting up the permissions.