innoveit/react-native-ble-manager #428: How to write byte array in write() method

marcosrubio3, 2018-12-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v "^6.5.4"
- react-native v: 0.55.4

### Expected behaviour

I need to pass data to my BLE peripheral in the following format (code in Java):
       byte[] bytes = new byte[6];
        bytes[0] = (byte) 0xAB;
        bytes[1] = (byte) 0;
        bytes[2] = (byte) 3;
        bytes[3] = (byte) 0xFF;
        bytes[4] = (byte) 0x71;
        bytes[5] = (byte) 0x80;

### Actual behaviour

I am having trouble to translate this code to an acceptable format for write() method of this library in react-native

### Stack trace and console log
I tried to create an array like var data = [stringToBytes('0xAB'), etc...] but I get the error com.facebook.react.bridge.ReadableNativeArray cannot be cast to java.lang.Double

I read that you need to cast these values to base64, but I do not know how to handle this case. Please help!

marcosrubio3, 2018-12-29:
I changed the code to:
  const data = [0xAB, 0, 3, 0xFF, 0x71, 0x80]
  console.log(data);

  BleManager.write(id, service, characteristic, data, maxByteSize)
    .then(() => {
      console.log('Write: ' + data);
    })
    .catch((error) => {
      console.log(error);
    });

However, my console log is:
(6) [171, 0, 3, 255, 113, 128]
Write failed

I retrieved the services and setTimeout() before calling the write method. Any ideas?
The characteristic I am trying to write to has properties Write and WriteWithoutResponse and shares the UUID with another characteristic with Notify property.

marcosinigaglia, 2019-01-09:
Hi @marcosrubio3 , you need to debug in your device, check the error detail.

lizujian, 2019-01-31:
@marcosrubio3  Let me show you code.
let command = ['AB', '00', '03', 'FF', '71', '80'];
let data = command.map(x => {return parseInt(x, 16);});
BleManager.write(id, service, characteristic, data, maxByteSize)
.then(() => {
console.log('Write: ' + data);
})
.catch((error) => {
console.log(error);
});

lizujian, 2019-01-31:
@marcosinigaglia by the way, I have a question why the example shows the tool 'covert-string'  doesn't   decode my hexString to digital array.
ie.
let command = 'AB0003FF7180';
const data = stringToBytes(command);
As result, the decoded data doesn't meet my need. The command string is split by '', however, I want to get 'AB','00','03'...like this to split and decode to array.

marcosinigaglia, 2019-01-31:
Hi @lizujian , I didn't add that piece in the documentation, I think is better to change it. Also I don't find that library documentation. In my project I always work with bytes directly so I never search for a library for the conversion.

closed 2019-02-21 by marcosinigaglia

priyanka-Sadh, 2020-04-15:
@Marcos-Costa did you fix the issue, I am facing the same "write failed", i am sending data in Array bytes in below example, also tried retrieve-services and settimeout.
`BleManager.write(id, service, characteristic,[10], maxByteSize)
.then(() => {
console.log('Write: ' + data);
})
.catch((error) => {
console.log(error);
});`

@lizujian please suggest something here, I am struggling since 1 week.

priyanka-Sadh, 2020-04-15:
@Marcos-Costa I tried writeWithoutResponse, that is working on iOS but still coming same "write failed" on android
@lizujian please suggest something here, I am struggling since 1 week.

viren-vii, 2021-06-20:
When I am using `stringToBytes` function, it fives error as 
```
    TypeError: (0, _convertString.default) is not a function. (In '(0, _convertString.default('somestring')', '(0, _convertString.default)' is an instance of Object)  
```
Do we have any solution here?

fmkeke233, 2022-09-30:
`

data = [1,2,3,4,4,5 ]
 BleManager.write(connectedDevice.id,server,characteristics, data, 180)
`
data should be a array ,don't use a ArrayBuffer ,