innoveit/react-native-ble-manager #429: I don't receive notification from BleManagerDidUpdateValueForCharacteristic

pierpieandthe, 2018-12-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v0.57.7
- iOS/Android v.8.0.0

### Expected behaviour

I should receive notification from a BLE device after writing a message to it.

### Actual behaviour

I write messages but receive no notification back.
I tested "manually" with nRF Connect tool and I receive a notification connecting to the right service and characteristic- so my code shall be wrong

### Steps to reproduce

Here is my code 

 ```
 componentDidMount() {
    AppState.addEventListener('change', this.handleAppStateChange);

    BleManager.start({showAlert: false});

    this.handlerDiscover = bleManagerEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral );
    this.handlerStop = bleManagerEmitter.addListener('BleManagerStopScan', this.handleStopScan );
    this.handlerDisconnect = bleManagerEmitter.addListener('BleManagerDisconnectPeripheral', this.handleDisconnectedPeripheral );

    bleManagerEmitter.addListener(
      'BleManagerDidUpdateValueForCharacteristic',
      ({ value, peripheral, characNotif, service }) => {
          console.log("received a notif")
          const data = bytesToString(value);
          console.log(`Received ${data} for characteristic ${characNotif}`);
      }
    );
```
and
```
  async connectAndSendMessage(peripheral) {
    
      var service = 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXX';
      var characNotif = 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXX';
      var characWrite = 'XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXX';
      var textToWrite = stringToBytes("XXXXXXXXX");

      await BleManager.connect(peripheral.id);
      await BleManager.retrieveServices(peripheral.id);
      await BleManager.startNotification(peripheral.id, service, characNotif);
      await BleManager.write(peripheral.id, service, characWrite, textToWrite)
  }
```

### Stack trace and console log
scan
got ble peripheral
stop scan

nothing more

marcosinigaglia, 2019-01-09:
Hi, try to add some delay between the operations. Check also the device log.

pierpieandthe, 2019-01-11:
I added some delay but also had problems with data conversion: message I was sending had extra 0, so my BLE device could not understand it

closed 2019-01-11 by pierpieandthe