innoveit/react-native-ble-manager #431: could not invoke Blemanager.scan

shengliZh, 2019-01-04:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Can scan

### Actual behaviour

Can't scan
Crash, could not invoke Blemanager.scan

### Steps to reproduce

1. open bluetooth
2. call Blemanager.scan([],120,true,option)

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-01-09:
Is BleManager not Blemanager, try to reinstall the library if the problem persist.

srikanthkyatham, 2019-01-15:
@shengliZh I presume you are using iOS, and trying to scan in the background, if so you need to do a filtered scan.

shengliZh, 2019-01-15:
@srikanthkyatham 
I am upgrading firmware, no filtered.

closed 2019-02-21 by marcosinigaglia