innoveit/react-native-ble-manager #432: Basic understanding

assafb81, 2019-01-08:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v0.57.8
- iOS v12.1.2

first let me say that the package is great.
I am trying to understand something.
I installed `BLE Peripheral Simulator` app on my android device and I m using the `Heart Rate Monitor` in it.

When I use `retrieveServices` I am getting a response with 3 characteristics:

1. characteristic: "2A37" -> with property "Notify"
2. characteristic: "2A38" -> with property "Read"
3. characteristic: "2A39" -> with property "Write"

So,
If I want to listen to the notifications,I need to register with characteristicUUID "2A37"?
If I want to write I need to do it with characteristicUUID "2A39"?

Thanks

assafb81, 2019-01-10:
Thanks for the quick reply.
I know that `pairing` and `bonding` are different this is why I asked the 1st question.
I will try to clarify my question - as I understand, correct me if I am wrong, `connect` is used to perform `pairing` between 2 devices.
- iOS: the OS is responsible to perform `bond` (according to OS parameters) - and we as developers don't have access to this api (including in SWIFT).
- Android: `bonding` is possible using `createBond`. Can I use `createBond` without `connect` or do I have to use `connect` prior to `createBond`?

marcosinigaglia, 2019-01-10:
You don't need to pair to a device to connect with ble. In iOS is the OS that manage the pairing and bonding process, in Android I think you can 'force' with `createBond`.

closed 2019-01-14 by assafb81