innoveit/react-native-ble-manager #436: Connect returns false positive on smasung devices

Keped, 2019-01-10:
### Version

Tell us which versions you are using:

- "react-native-ble-manager": "^6.5.4"
- "react-native": "^0.52.0"
- Android 8.0.0 (Samsung S8)

### Expected behaviour

on every other device, it is enough to call retrieveServices after .connect resolves successfully. after retrieveServices resolves too, I can read a characteristic.

### Actual behaviour

On Samsung S8 devices running Android 8.0.0 the .connect resolves "successfully"  but retrieveServices behaves inconsistently - sometimes it works immediately, but more times it will return with "device has disconnected" and will work only after 2-3 tries (with some wait between each) 

### Steps to reproduce

1. discover, connect and try to read from a ble peripheral with a Samsung S8 

### Stack trace and console log
 "device has disconnected"

marcosinigaglia, 2019-01-10:
Hi, are you using the last version?

Keped, 2019-01-10:
@marcosinigaglia thanks, version 6.5.4 ... I now see you went to 6.5.6 yesterday :) 🥇 ,do you think upgrading can solve it?

Keped, 2019-01-14:
(tested on previously problematic devices) Apparently fixed with version 6.5.6

closed 2019-01-14 by Keped

Keped, 2019-01-15:
@kyle-ssg yes, our QA guy is positive he is not seeing the problem anymore on samasung S8 Android 8.0.0.