innoveit/react-native-ble-manager #437: Not getting full BLE peripheral name

ohjeyong, 2019-01-11:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.2
- react-native v0.57.4
- iOS/Android I have three phones, Samsung galaxy 8/9, LG G5 and confirm the problem on every three phones. 

### Expected behaviour

When discover peripheral, peripheral.name has its own name.

### Actual behaviour
I have a lot of scooters with built in BLE. Scooter's BLE peripheral name is 'MiscooterXXXX' which XXXX is random number. Most of the time, peripheral.name has its own name for example Miscooter5521. But sometime, (I guess one or two out of ten) peripheral.name show only 'Misc'.
I have a lot of scooters and three phones and it is weird that this problem appears randomly on random scooter, random phones.
If I try to connect 'Misc' peripheral, then it is connected successfully. However, what I try to do is when there are multiple scooters tuned on, show their peripheral name list to user in react-native app. When user select the peripheral name, then the scooter which has selected peripheral name will be connected.(Scooter's XXXX number is printed on scooter so user can distinguish them by peripheral name.) The problem is, if there are multiple 'Misc' in list, user can't distinguish scooters.

### Steps to reproduce

1. scan
2. list the peripheral name
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

ohjeyong, 2019-01-13:
> Hi @ohjeyong , is not a library bug. I think using peripheral name is not a good way to identify a device.

@marcosinigaglia 
First time, I considered mac address to identify a device but mac address is not consistent on ios, so I thought using peripheral name is a solution. Is there any way to identify a device? Thanks.

srikanthkyatham, 2019-01-15:
In iOS the platform gives unique UUID for the iBeacon

marcosinigaglia, 2019-01-15:
Try to look in the `advertising` field if you have other informations.

closed 2019-06-18 by marcosinigaglia