innoveit/react-native-ble-manager #438: Device is not connected

harshit275, 2019-01-14:
### Version

Tell us which versions you are using:

- react-native-ble-manager v.6.5.6
- react-native v.0.57.8
- Android v.27.0.3

### Expected behavior

When trying to connect to VEEPEAK OBD 2 dongle and call retrieveServices method it should give all the services PIDs and characteristics.

### Actual behavior

When I am trying to connect my react native app to the VEEPEAK OBD2 dongle it gets connected but within a second it gets disconnected. Before connecting I am also creating a bond with the device. The bond created successfully but still after getting connected it gets disconnected.
Previously I was using the v.6.5.4 of the library that time also it is giving the same behavior and after upgrading to the latest v.6.5.6 still the things are same for me.
The device I am trying to connect is a Veepeak OBDCheck BLE OBD2 Bluetooth Scanner which works on Bluetooth 4.0, link for this device is:
[https://www.amazon.com/Veepeak-OBDCheck-Bluetooth-Diagnostic-Compatible-x/dp/B073XKQQQW](url)

Also please check the code below:
`BleManager.createBond(peripheral.id).then(() => {
                            console.log('createBond success or there is already an existing one');
                            BleManager.connect(peripheral.id).then(() => {
                                    let peripherals = this.state.peripherals;
                                    let p = peripherals.get(peripheral.id);
                                    if (p) {
                                        p.connected = true;
                                        peripherals.set(peripheral.id, p);
                                        this.setState({ peripherals });
                                    }
                                    console.log('Connected to ' + peripheral.id);
                                    setTimeout(() => {
                                        BleManager.retrieveServices(peripheral.id).then((peripheralData) => {
                                            console.log('Retrieved peripheral services', peripheralData);
                                            setTimeout(() => {
                                                BleManager.read(peripheral.id, serviceUUID, characteristicUUID).then((readData) => {
                                                    console.log('Read: ' + readData);
                                                }).catch((error) => {
                                                    console.log(error);
                                                });
                                            });
                                        }).catch((error) => {
                                            console.log('Error: ' + error);
                                        });
                                    });
                                }).catch((error) => {
                                    console.log(error);
                                });
                        }).catch(() => {
                            console.log('fail to bond');
                        })`

### Steps to reproduce

1. Scan for peripherals
2. List all peripherals
3. Create a bond with the peripheral
4. Connect to the peripheral
5. Retrieve services of the peripheral

### Stack trace and console log

Create bond success or there is already an existing one
Connected to 8C:XX:XX:XX:XX:CD
Disconnected to to 8C:XX:XX:XX:XX:CD
Error: Device is not connected

marcosinigaglia, 2019-01-15:
Hi, can you try the BLE scanner app?

harshit275, 2019-01-15:
Hi, I have tried using the BLE scanner app and it works quite okay, is able to connect as well as fetch PIDs. Perhaps, I am missing something in the usage of this library (react-native-ble-manager) and if you can point me to that, it would be helpful.

marcosinigaglia, 2019-01-15:
Try to connect without calling `createBond` before or to adding some delay beetween the `connect` and `retrieveServices` calls.

harshit275, 2019-01-16:
I have tried both of these approaches but the still getting the same result.

harshit275, 2019-01-16:
NO. I first let the scan to finish after that I connect it to the device.

marcosinigaglia, 2019-01-16:
So I have no idea, you have to debug the error on the device.

harshit275, 2019-01-16:
You have talked about that BLE Scanner app in your previous comment, does that app also works on this library (react-native-ble-manager).

folofse, 2019-03-13:
I had the same problem on my Pixel 2, but after updating to Android 9 build: PQ2A.190305.002 the problem seems to have disappeared. I should do some more testing but it might been solved.

folofse, 2019-03-13:
I have the same problem, it works the first time I connect, but after disconnect and connect again a number of times its failing. I can see that `BLEManager.isConnected(peripheralId)` returns true but `BLEManager.write(peripheralId ...` returns error "Device is not connected" By removing `/*if (!isConnected()) {
			callback.invoke("Device is not connected", null);
			return;
		}*/` on line 679 in Peripheral.java it manage to get it to work, so the function isConnected in Peripheral.java is not using the same functions as `BLEManager.isConnected(peripheralId)` to control if the peripheral is connected. I'm not yet sure if the removal of the if-statement might have negative effects.

closed 2019-06-18 by marcosinigaglia

maartenvandillen, 2021-12-21:
I have the same problem. It starts at random after some app runs. I suspect it has something to do with the underlying Bluetooth layer of Android. It gets really weird when I completely disable Bluetooth from the phone settings, the connection still exists and I can still operate my peripheral. Even after killing my app I can still connect to it while Bluetooth is turned off. I suspect this connection keeps existing even after the app is restarted. Only rebooting my phone (Samsung Galaxy S20+) solves this.

vishal980, 2022-09-29:
isconnectable false device not connecting. isconnectable false how to connect device.

maartenvandillen, 2022-09-29:
I got it all working now, the library works fine. When React Native hot reloads the app the connection persists so then it gets confused. Solved by disabling Bluetooth from the phone settings menu (not from the quick pull system down menu, that button is pure evil on Samsung devices) to disconnect the device.

If you device reports isConnectable false then there may already be a current connection or maybe something is wrong in the firmware.