innoveit/react-native-ble-manager #439: How call BleManager.start few times

VitamineC, 2019-01-14:
- react-native-ble-manager v6.0.2
- react-native v0.5.4

Hello everyone, 
I need get enable to call BleManager.start few times on Ios. 
I understand that I should call it only once, but I use 
https://github.com/Pilloxa/react-native-nordic-dfu
And this library particularly destroy ble-instance, and once opportunity that can help me, that enable recreate bleInstance.

marcosinigaglia, 2019-01-15:
Hi @VitamineC , i think you have to add a new method to force the initialization of the library, now this situation is not expected.

shengliZh, 2019-01-21:
Hi,
I only need to call it once, I am also using the library of react-native-nordic-dfu and successfully upgrading the firmware.
Send the command to let the firmware enter the dfu mode and then directly scan the dfu device.
It won't destroy ble-instance.

VitamineC, 2019-01-21:
@shengliZh 
I have this problem only on IOS. On android all works nice. 

I forked this library and allow (only for ios) recall BleManager.start() few times without any problems.

closed 2021-02-18 by marcosinigaglia