innoveit/react-native-ble-manager #441: IOS | how to read data more than 512 bytes using BLE react native. now we can read only 512 bytes of data

yoginitheesh, 2019-01-15:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-01-15:
Hi, you have to split your data and handle this "queue" by your app.

yoginitheesh, 2019-01-15:
@marcosinigaglia  Scenarios: 1- After sending write request m sending one read request ,that time m getting only 512 bytes of data. but actual size is more than 512 bytes. again no need to send read request to read remaining data rite? . In android its was working fine. In IOs m getting this issue. Any suggestions?

marcosinigaglia, 2019-01-15:
I think Is better to use the notification so you can have multiple notifications after your write.

yoginitheesh, 2019-01-15:
Thanks @marcosinigaglia. Let me check.

sushanthNiveus, 2019-01-22:
@marcosinigaglia Hi Marc. Facing the same issue here. Sending the same data to both Android and iOS. In Android complete data is read properly but in iOS data is truncated. If the issue is on BLE peripheral side, data wouldn't have sent to Android also.  Any idea?
[image]

marcosinigaglia, 2019-01-22:
Hi @sushanthNiveus , is not an issue, is an OS specification, you have to hadle it manually.

closed 2019-06-18 by marcosinigaglia