innoveit/react-native-ble-manager #444: Peripherals scanned has no values in serviceUUID

antring, 2019-01-17:
### Version
- react-native-ble-manager v0.57.8
- react-native v6.5.6
- Android v9 (same problem present before upgrade to 9)

### Expected behaviour
Scanned devices returning service UUIDs which in turn can be used to filter during scanning to remove unwanted devices.

### Actual behaviour
When scanning for devices, the devices do not return any UUIDs resulting in not being able to filter the scan. 

### Steps to reproduce

1. Scan for devices
2. Print peripheral info to console

marcosinigaglia, 2019-01-21:
Hi @antring , you have to debug in your device.

antring, 2019-01-22:
This is the content in console while the app is scanning. 
No errors or warnings are printed. The informations is primarily info that I print to the console. 
As you can see the serviceUUIDS list are empty. 
If I connect to the device and scan, the list is populated according to what the device is configured to broadcast. 

01-22 13:47:21.851 19148 19538 I ReactNativeJS: (!) Module initialized
01-22 13:47:21.851 19148 19538 I ReactNativeJS: (!) User accepted permissions
01-22 13:47:21.852 19148 19538 I ReactNativeJS: (!) Bluetooth enabled or user confirmed
01-22 13:47:34.319 19148 19538 I ReactNativeJS: (!) Starting scan
01-22 13:47:35.245 19148 19538 I ReactNativeJS: 'Found: ', { advertising:
01-22 13:47:35.245 19148 19538 I ReactNativeJS:    { serviceUUIDs: [],
01-22 13:47:35.245 19148 19538 I ReactNativeJS:      isConnectable: true,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:      txPowerLevel: -2147483648,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:      manufacturerData:
01-22 13:47:35.245 19148 19538 I ReactNativeJS:       { bytes:
01-22 13:47:35.245 19148 19538 I ReactNativeJS:          [ 12,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            9,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            110,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            82,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            70,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            32,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            67,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            111,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            110,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            110,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            101,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            99,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            116,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:            0 ],
01-22 13:47:35.245 19148 19538 I ReactNativeJS:         data: 'DAluUkYgQ29ubmVjdAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=',
01-22 13:47:35.245 19148 19538 I ReactNativeJS:         CDVType: 'ArrayBuffer' },
01-22 13:47:35.245 19148 19538 I ReactNativeJS:      localName: 'nRF Connect' },
01-22 13:47:35.245 19148 19538 I ReactNativeJS:   rssi: -46,
01-22 13:47:35.245 19148 19538 I ReactNativeJS:   id: 'FC:B3:54:8A:7B:A7',
01-22 13:47:35.245 19148 19538 I ReactNativeJS:   name: 'nRF Connect' }
01-22 13:47:39.324 19148 19538 I ReactNativeJS: (!) Scanning stopped
01-22 13:47:39.467 19148 19538 I ReactNativeJS: (!) Results: 1

Is it perhaps something I do while adding filtering to the scanning method?
I have tried the following: 
`BleManager.scan(['4320'], 5, true)`
`BleManager.scan(['00004320-0000-1000-8000-00805f9b34fb'], 5, true)`
And without '-' but that returns an error. 
The UUID is correct in what is displayed in the nRF Connect app. 
Scanning with filtering returns no results.

antring, 2019-01-22:
Just to confirm that I have understood correctly. To filter out unwanted devices I should connect to all available devices, retrieve services, disconnect and then add the relevant devices to my map to display for the user?

antring, 2019-01-22:
The serviceUUID is correct, and I identified the issue. 
My device was not broadcasting the UUID in the advertising packet, which in turn made it not visible when filtering the scan...
I added the service to the advertising packet and now it works as expected. 
Thank you for the assistance!

closed 2019-01-22 by marcosinigaglia

yang200258, 2019-08-02:
Hi,same issues as you,how added the service to the advertising packet?
thanks a lot!

antring, 2019-10-18:
> Hi,same issues as you,how added the service to the advertising packet?
> thanks a lot!

Sorry for the late reply.....
Adding a service to the advertising packet is something done in firmware on the device.

Manmapi, 2023-01-31:
> The serviceUUID is correct, and I identified the issue. My device was not broadcasting the UUID in the advertising packet, which in turn made it not visible when filtering the scan... I added the service to the advertising packet and now it works as expected. Thank you for the assistance!

This resolve my problem, thanks a lot