innoveit/react-native-ble-manager #448: Java exception in 'NativeModules' java.lang.NoClassDefFoundError: android/bluetooth/le/ScanRecord

shengliZh, 2019-01-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.6
- react-native v0.56
- iOS/Android Android4.4.2

### Expected behaviour

Open app will crash

### Actual behaviour

Java exception in 'NativeModules' java.lang.NoClassDefFoundError: android/bluetooth/le/ScanRecord

### Steps to reproduce

1.  Open app
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-01-21:
Hi, check your project api level.

shengliZh, 2019-01-22:
Hi, @marcosinigaglia 
This is my project api level, my phone OS Version is 4.4.2.
[image]

lvlrSajjad, 2019-02-10:
Same here, min SDK version is 19

yangxy112683, 2019-02-18:
same here, min SDK version is 19, RN version is ^0.57.8;

owen12ab, 2019-04-17:
Same here, min SDK version is 19 and RN version is 0.56.0, my android device os is 4.4.4. Any update on this issue?

JackClown, 2019-04-24:
Catch the same problem.I find that `android.bluetooth.le.ScanRecord` is added in API level 21.

marcosinigaglia, 2019-06-25:
Hi @shengliZh , can you try the latest release?

closed 2019-06-28 by marcosinigaglia