innoveit/react-native-ble-manager #450: Unknowm argument type '_Bool' in method .... Extend RCTConvert to support this type.

ducNgbk, 2019-01-23:
Hi, I am facing this problem

### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.0
- react-native v0.57.7
- iOS 12.1

### Expected behaviour

`disconnect()`works fine

### Actual behaviour

`disconnect()` cause this error 

> Unknowm argument type '_Bool' in method .... Extend RCTConvert to support this type.

### My workaround
The original code
```
RCT_EXPORT_METHOD(disconnect:(NSString *)peripheralUUID force:(bool)force callback:(nonnull RCTResponseSenderBlock)callback)
{
...
}
```
I changed to
```
RCT_EXPORT_METHOD(disconnect:(NSString *)peripheralUUID force:(BOOL)force callback:(nonnull RCTResponseSenderBlock)callback)
{
...
}
```
then it works

Hope this might help.
Thanks for your awesome lib

marcosinigaglia, 2019-01-23:
Hi, I think is a problem from the latest PR.

marcosinigaglia, 2019-01-23:
Fixed

closed 2019-01-23 by marcosinigaglia