innoveit/react-native-ble-manager #451: installation setup for ios

kirhim, 2019-01-23:
I am really excited about this library but I am confused with the installation guide.

I followed the steps below and it seems that the installation was successful.

1.  npm i --save react-native-ble-manager
2. react-native link react-native-ble-manager
3. import BleManager from 'react-native-ble-manager'

What about from xcode? I don't have to do anything? 
I don't have to add `CoreBluetooth.framework` in the` Linked Frameworks and Libraries ` ?

By the way tried with/without adding `CoreBluetooth.framework` but couldn't scan BLE devices from my phone.

ducNgbk, 2019-01-23:
Hi, by default `react-native link react-native-ble-manager` will do **almost** everything
So you have to check by follow `Link Native Library Manually` section if build error
Btw, you don't need to care about `CoreBluetooth.framework` 

If you still have this trouble, please give more detail

kirhim, 2019-01-23:
@ducNgbk thank you very much for the reply. I don't get an error but I can't scan BLE devices. I am following [https://eysermans.com/post/scanning-for-bluetooth-devices-with-react-native](url) this link. 

All I did was change the class name and copy this code in my App.js. 
`import React, { Component } from 'react';
import { 
    AppRegistry,
    ListView,
    View, 
    Text, 
    Button } from 'react-native';

export default class App extends Component {
    constructor(props){
        super(props);

        const dataSource = new ListView.DataSource({rowHasChanged: (r1, r2) => r1 !== r2});
        this.state = {
            dataSource: dataSource.cloneWithRows(['row 1', 'row 2', 'row 3', 'row 4', 'row 5', 'row 6', 'row 7', 'row 8'])
        };
    }

    startScanning() {
        console.log('start scanning');
    }

    render() {
        return (
            <View style={{padding: 50 }}>
                <Text>Bluetooth scanner</Text>
                <Button onPress={() => this.startScanning()} title="Start scanning"/>

                <ListView
                    dataSource={this.state.dataSource}
                    renderRow={(rowData) => <Text>{rowData}</Text>}
                />
            </View>
        );
    }
}`

If you could give me any comments or advise it would be really helpful. Thanks in advance! :)

closed 2019-01-24 by kirhim