innoveit/react-native-ble-manager #455: error  writing value

kirhim, 2019-01-25:
I am trying to write value so that I could turn on my led using HM-10 module.

I did the following steps:
`import { stringToBytes } from 'convert-string';`
`const data = stringToBytes('on');`

```
writeValue(){
  BleManager.write('831827FC-8553-9E32-9DB0-BCCE941AC99F', 'FFE1', data)
  .then(() => {
    // Success code
    console.log('Write: ' + data);
  })
  .catch((error) => {
    // Failure code
    console.log(error);
  });
}
```

When I call this method, I get an error saying `JSON value'(111,110)' of type NSMutableArray cannot be converted to NSString`

What I am trying to do is pass `on` as a data to the ble module. Could anyone help?

marcosinigaglia, 2019-01-25:
Hi @kirhim , please check the documentation, you are missing a parameter.

kirhim, 2019-01-26:
@marcosinigaglia Yep, my mistake thank you so much 👍

closed 2019-06-18 by marcosinigaglia

ankushdec1993, 2020-02-25:
What parameter did you add?

ankushdec1993, 2020-02-25:
@marcosinigaglia

marcosinigaglia, 2020-02-25:
https://github.com/innoveit/react-native-ble-manager#writeperipheralid-serviceuuid-characteristicuuid-data-maxbytesize

ankushdec1993, 2020-02-25:
Im sending the data like that shred in the image below

ankushdec1993, 2020-02-25:
But its still not working

ankushdec1993, 2020-02-25:
@marcosinigaglia  now i am sending the "on" as a parameter using convert string npm,
Im getting value's length is invalid i'm giving the length also

ankushdec1993, 2020-02-25:
Hi Marco,

Can you send me the format of data that i should be sending like an
example(not the complete code only the data that we send)?