innoveit/react-native-ble-manager #456: Bluetooth Devices not listing in android

Sanjeevs1995, 2019-01-28:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v0.57.8
- Android v.8.0

### Expected behavior

ESP 32 Bluetooth module should get listed in the application

### Actual behavior

The device does not get listed. 

But it shows in the OS Bluetooth listing UI

The application works fine for all android version<8.

marcosinigaglia, 2019-01-29:
Hi @Sanjeevs1995 , Android 8 have a lot of issue with the bluetooth, try to google and find a solution, I don't thinks is a library problem but a OS problem.

Sanjeevs1995, 2019-01-29:
But while using native code the device gets listed

I followed the steps mentioned in android native developer site

https://developer.android.com/guide/topics/connectivity/bluetooth#java

marcosinigaglia, 2019-01-29:
So figure out the difference.

closed 2019-06-18 by marcosinigaglia