innoveit/react-native-ble-manager #457: How to use the write() method correctly

harshit275, 2019-01-28:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.4
- react-native v0.57.8
- Android v.27.0.3

### Expected behavior

Well, I should get my result successfully once I fired the PID to my BLE device using the `write()` method.
This is PID 05 in mode 01, and can be requested as follows: 
>01 05 
The response will be of the form: 
41 05 7B 
The 41 05 shows that this is a response to a mode 1 request for PID 05, while the 7B is the desired data. 

### Actual behavior

First, let me show you the `write()` method which I am executing after successful retrieval of the services.
`setTimeout(() => {
  BleManager.startNotification(peripheral.id, serviceUUID, characteristicUUIDN)
    .then(() => {
      console.log('Started notification on ' + peripheral.id);
      setTimeout(() => {
        var temp = BleManager.write(peripheral.id, serviceUUID, characteristicUUIDW, pids)
          .then(() => {
            console.log('Writed PID: ' + pids);
          }).catch((error) => {
            console.log('Error while writing: ' + error);
          });
        console.log(temp);
      }, 500);
    }).catch((error) => {
      console.log('Error while notification: ' + error)
    });
}, 900);`

Now as I am using the `write()` method with response so here is the response as Promise object which I am getting after execution of write method:
>Promise {_40: 0, _65: 0, _55: null, _72: null}

And here is the notification I am getting everytime from my BLE device:
>{
	characterstic: "000fff1-0000-XXXX-XXXX",
	peripheral: "8C:XX:XX:XX:XX:CD",
	service: "000fff0-0000-XXXX-XXXX",
	value: [63, 13, 13, 62]
}

It doesn't matter what data I pass everytime I get this very same result from my device. I am not sure wheather this is an error or some kind of result or am I missong something while writing the PID.
The device which I am using for testing my code is a Freematics OBD-II Emulator.

Also, can you please give me an example of `write()` method, I mean what's the correct way to use this method and also can you please elaborate on the output which we get after its successful execution and what kind of output we get in the form of notification.

marcosinigaglia, 2019-01-29:
Hi @harshit275 , the write method don't have value in response. I think you have to read or to handle some notification from the device.

harshit275, 2019-01-29:
Hey, @marcosinigaglia thanks for your reply.
I understand that the `write()` method doesn't give me any value but that's where I am facing difficulties.
I have handled notifications in the way as you have mentioned in the example folder, you can also have a look in my code snippet mentioned above. Please give me an example as to how this `startNotification()` thing works in this and how can we handle notifications.

marcosinigaglia, 2019-01-29:
You need to handle the `BleManagerDidUpdateValueForCharacteristic` event, check the documentation.

harshit275, 2019-01-29:
Yea man I have done that:
`handleUpdateValueForCharacteristic(data) {
        console.log('Received data from ' + data.peripheral + ' characteristic ' + data.characteristic, bytesToString(data.value));
    }`
Still not  getting any desired result.
Getting this same notification everytime:
>{
characterstic: "000fff1-0000-XXXX-XXXX",
peripheral: "8C:XX:XX:XX:XX:CD",
service: "000fff0-0000-XXXX-XXXX",
value: [63, 13, 13, 62]
}

Can you please read my query carefully and understand it what I am trying to ask.
Many thanks in advance.

marcosinigaglia, 2019-01-29:
The write method is simple and you are using it correctly, also you are reading data from the peripheral so I don't understand the problem you have. You have to check the device documentation or contact the vendor support.

harshit275, 2019-01-31:
@lizujian Thanks for your comment. As I have already mentioned above I have already written `handleUpdateValueForCharacteristic` method for handling responses and all. But in my case, I am getting the same every time no matter what data I pass the response is same.
>{
characteristic: "000fff1-0000-XXXX-XXXX",
peripheral: "8C:XX:XX:XX:XX:CD",
service: "000fff0-0000-XXXX-XXXX",
value: [63, 13, 13, 62]
}
Please have a look at this response and comment if this sounds anything to you.
Many thanks in advance.

closed 2019-06-18 by marcosinigaglia

yoonachen, 2021-12-14:
Hey, do you solve this problem?
I have the same issue...