innoveit/react-native-ble-manager #458: Android | One plus 5 having some connection not able to list near by bluetooth devices

yoginitheesh, 2019-01-29:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

yoginitheesh, 2019-01-29:
@marcosinigaglia .

srikanthkyatham, 2019-01-30:
how are you scanning. 
If you could paste some code. 
Did you check whether the bluetooth is on. Where the permissions for bluetooth granted.
Did you have the correct permissions in the AndroidManifest file

yoginitheesh, 2019-01-31:
@srikanthkyatham . Now its working fine after adding     **<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>**

yoginitheesh, 2019-01-31:
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

closed 2019-01-31 by yoginitheesh