innoveit/react-native-ble-manager #460: Crash on writeWithoutResponse

mahyali, 2019-02-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.5.2
- react-native v0.57.8
- Android v28

### Expected behaviour

writeWithoutResponse should send binary data to the ble device

### Actual behaviour

I get a crash (red screen) containing an error saying "Malformed calls from JS: field sizes are different"

### Steps to reproduce
`
        await BleManager.writeWithoutResponse(
                this.state.device.id,
                this.UUID_SERVICE,
                this.UUID_TX,
                binaryMessage)
                .catch(error => alert(error));
`
Calling that function causes the crash. 

### Stack trace and console log
[image]

```
```

marcosinigaglia, 2019-02-01:
Hi, do you have the same error using `write`?

mahyali, 2019-02-02:
I think I figured it out. Using a plain array works but not a typed Array like Uint8Array for some reason.

closed 2019-02-21 by marcosinigaglia