innoveit/react-native-ble-manager #461: iOS | BleManager.retrieveServices won't  works. for reconnecting

yoginitheesh, 2019-02-01:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1.
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-02-01:
?

vidhyakrishna11, 2019-02-28:
@yoginitheesh Did you find a solution to this issue?
I am facing the same. Kindly let me know

vidhyakrishna11, 2019-02-28:
@marcosinigaglia 

* react-native-ble-manager - 6.5.1
* react-native - 0.55.4
* iOS  - 12.1.4

## Expected behaviour
It should read from characteristics after pairing 

### Actual behaviour
after pairing it shows this error while reading from characteristics - "Peripheral not found or not connected"

### Steps to reproduce
1.Try to pair with BLE device
2.Once pairing is done, try to read  from any characteristics.
3.Every first time it pass, but when you try next time it fails with the given error/ expectation.
4.This keeps on repeating until you restart the app .

closed 2019-06-18 by marcosinigaglia

soadtech, 2021-09-07:
Possible Unhandled Promise Rejection (id: 1):
"Peripheral not found or not connected"