innoveit/react-native-ble-manager #463: Example code not working

xwilluhn, 2019-02-05:
### Version

Tell us which versions you are using:

    "react": "16.6.3",
    "react-native": "0.58.3",
    "react-native-ble-manager": "^6.6.1",

mac (simulator)

### Expected behaviour

it should scan ble devices and display them

### Actual behaviour

after click 'Scan Bluetooth', the console.log output 'start scanning', a few seconds later, 'Scan is stopped'

### Steps to reproduce

react-native init myApp
npm i --save react-native-ble-manager
npm install
react-native link react-native-ble-manager
react-native run-ios

### Stack trace and console log

no error. just not scan any devices

```
```

closed 2019-02-05 by marcosinigaglia

xwilluhn, 2019-02-07:
> Hi, you need a real device, is not working on the simulator.

Hi Marco,

Thanks for the feedback. I got the app running on my iPhone X, still the
same behavior. I click on, a few seconds later, turn it off automatically.
I saw TCP Conn failed

Any suggestions what should I do?

I am writing a simple app with react native to scan any Bluetooth low
energy devices nearby and display them and their data.

Thanks a lot,
Best, Xia

marcosinigaglia, 2019-02-07:
TCP conn error is common, I don't think is that the problem, you have to debug.

xwilluhn, 2019-02-07:
Hi Marco,

thanks. In the meantime, I got the example working!!
The TCP Conn has something to do with the react mative debugging thing

Anyhow, think for writing and maintthe library!

Best,
Xia