innoveit/react-native-ble-manager #465: Meaning of Services and Characteristics

tal987, 2019-02-07:
### Version

Tell us which versions you are using:

- react-native-ble-manager v?.?.?
- react-native v?.?
- iOS/Android v.?

### Expected behaviour

Hi, I know its beginner question, recently im facing with bluetooth in react native.
What I didn't find is explanation about:
1. what are the services and the characteristics means?
2. How can I know what each service is doing?
3. How can I understand the meaning of the characteristics?
4. Now to convert String to character?

Thank you for the answers

marcosinigaglia, 2019-02-07:
https://medium.com/@zpcat/how-bluetooth-le-works-link-layer-b18475250259

tal987, 2019-02-07:
@marcosinigaglia Thank you for the answer but I didnt see any information about the services or the characteristics, the article is only one page that explains the layers

MikeShi42, 2019-02-08:
Services and Characteristics are more general to BLE, probably a better question suited to somewhere like StackOverflow.

This might help: https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch04.html

closed 2019-06-11 by marcosinigaglia