innoveit/react-native-ble-manager #467: Write without call startNotification before

tal987, 2019-02-11:
Just a question of beginner:

Do I have to call startNotification function before I use the function Write?
I know I have to call retrieveServices function before Write but what about startNotification...

Thanks in advance

marcosinigaglia, 2019-02-18:
Hi @tal987 , `startNotification` is to read data from the device so depends, probably is not necessary.

tal987, 2019-02-18:
@marcosingaglia so if startnotification read the data from the device so why do we mend the read function? What is the difference between the read function and the startnotification function?

marcosinigaglia, 2019-02-18:
Using `startNotification` you start to read the data that the device send to you depends on its specification. Using `read` you ask a value.

tal987, 2019-02-18:
@marcosinigaglia when I use the read fucnction, from the readMe file I understand it return promise but the data inside the promise is in type String or Byte array?

marcosinigaglia, 2019-02-19:
Is an array, check the example in the readme.

closed 2019-06-18 by marcosinigaglia