innoveit/react-native-ble-manager #471: invalid index.d.ts in 6.6.2

omatrot, 2019-02-19:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- React Native Environment Info:
    System:
      OS: macOS 10.14.3
      CPU: (4) x64 Intel(R) Core(TM) i5-6500 CPU @ 3.20GHz
      Memory: 125.50 MB / 16.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 8.11.2 - /usr/local/bin/node
      Yarn: 1.10.1 - /usr/local/bin/yarn
      npm: 6.8.0 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.1, macOS 10.14, tvOS 12.1, watchOS 5.1
      Android SDK:
        API Levels: 14, 22, 23, 24, 25, 26, 27, 28
        Build Tools: 22.0.1, 23.0.1, 25.0.0, 25.0.1, 25.0.2, 26.0.0, 26.0.1, 26.0.2, 27.0.3, 28.0.2, 28.0.3
        System Images: android-23 | Intel x86 Atom
    IDEs:
      Android Studio: 3.3 AI-182.5107.16.33.5264788
      Xcode: 10.1/10B61 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.6.0 => 16.6.0 
      react-native: 0.57.4 => 0.57.4 

### Expected behaviour

index.d.ts in 6.6.1 has:
```
export function getConnectedPeripherals(serviceUUIDs: string[]): Promise<any[]>
```

while in 6.6.2 it becomes 

```
export function getConnectedPeripherals(serviceUUIDs: string[]): Promise<void>
````

which breaks existing code.

### Actual behaviour

The typescript code used in 6.6.1 fails to compile in 6.6.2.

marcosinigaglia, 2019-02-19:
Hi @omatrot , the file is not changed.

omatrot, 2019-02-19:
Strange... I cloned a repo on a new computer and fetched dependencies and boum. What could have happened?

marcosinigaglia, 2019-02-21:
I don't know, I don't use typescript, if the ts file is wrong please make a PR.

omatrot, 2019-02-21:
PR done.

closed 2019-02-21 by omatrot