innoveit/react-native-ble-manager #473: Characteristic not foun

amjalca, 2019-02-20:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v.58.8
- iOS/Android v.6.0.2

### Expected behaviour

Tell us what should happen

### Actual behaviour

Tell us what happens instead

### Steps to reproduce

1. 
2.
3.

### Stack trace and console log

Hint: it would help a lot if you enable the debugger ("Pause on exceptions" in the "Source" panel of Chrome dev tools) and spot the place where the error is thrown

```
```

marcosinigaglia, 2019-02-20:
Did you call `retrieveServices` before? Is the characteristic in the return data?

amjalca, 2019-02-20:
Yes, I call retrieveServices. I have several characteristic for read info of beacon
```
'Characteristics ---> ', { characteristics: 
02-20 14:11:04.550 31853 31894 I ReactNativeJS:    [ { properties: { Write: 'Write', Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a00',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '1800' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a01',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '1800' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a04',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '1800' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a29',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a24',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a25',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a27',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a26',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { properties: { Read: 'Read' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a28',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { descriptors: [ { value: null, uuid: '2902' } ],
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        properties: 
02-20 14:11:04.550 31853 31894 I ReactNativeJS:         { Notify: 'Notify',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:           Write: 'Write',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:           WriteWithoutResponse: 'WriteWithoutResponse' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '7f280002-8204-f393-e0a9-e50e24dcca9e',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '7f280001-8204-f393-e0a9-e50e24dcca9e' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { descriptors: [ { value: null, uuid: '2902' } ],
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        properties: { Notify: 'Notify', Write: 'Write' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '2a08',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '7f280001-8204-f393-e0a9-e50e24dcca9e' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { descriptors: [ { value: null, uuid: '2902' } ],
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        properties: 
02-20 14:11:04.550 31853 31894 I ReactNativeJS:         { Notify: 'Notify',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:           Write: 'Write',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:           WriteWithoutResponse: 'WriteWithoutResponse' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        characteristic: '7f280005-8204-f393-e0a9-e50e24dcca9e',
02-20 14:11:04.550 31853 31894 I ReactNativeJS:        service: '7f280004-8204-f393-e0a9-e50e24dcca9e' } ],
02-20 14:11:04.550 31853 31894 I ReactNativeJS:   services: 
02-20 14:11:04.550 31853 31894 I ReactNativeJS:    [ { uuid: '1800' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { uuid: '1801' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { uuid: '180a' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { uuid: '7f280001-8204-f393-e0a9-e50e24dcca9e' },
02-20 14:11:04.550 31853 31894 I ReactNativeJS:      { uuid: '7f280004-8204-f393-e0a9-e50e24dcca9e' } ]
```
In the console, i can see

```
02-20 14:04:27.320 31853 31894 I ReactNativeJS: 'Started notification on AC:23:3F:A0:04:93'
02-20 14:04:27.850 31853 31894 I ReactNativeJS: 'In Read -> ', 'Characteristic 7f280002-8204-f393-e0a9-e50e24dcca9e not found.'
02-20 14:04:27.980 31853 31894 I ReactNativeJS: 'Write: ', [ [Getter/Setter],
02-20 14:04:27.980 31853 31894 I ReactNativeJS:   [Getter/Setter],
02-20 14:04:27.980 31853 31894 I ReactNativeJS:   [Getter/Setter],
02-20 14:04:27.980 31853 31894 I ReactNativeJS:   [Getter/Setter] ]
02-20 14:04:27.980 31853 31894 I ReactNativeJS: 'Received data from AC:23:3F:A0:04:93 characteristic 7f280002-8204-f393-e0a9-e50e24dcca9e', [ 69, 82, 82, 32, 67, 77, 68 ]
```

marcosinigaglia, 2019-02-21:
Hi, the characteristic is not readable.

closed 2019-02-21 by amjalca

samsun009, 2020-04-30:
> Yes, I call retrieveServices. I have several characteristic for read info of beacon
>
> ```
> 'Characteristics ---> ', { characteristics:
> 02-20 14:11:04.550 31853 31894 I ReactNativeJS:    [ { [...]

can you share your code used for write? because I have error while write characterstic not found.