innoveit/react-native-ble-manager #474: I need help to troubleshoot inconsistent data randomly received

omatrot, 2019-02-21:
### Version

Tell us which versions you are using:

- react-native-ble-manager v6.6.2
- react-native v0.57.4
- iOS/Android v.12.1.4/7.1.1

### Expected behaviour

Conistent data received from an oximeter IOT device.

### Actual behaviour

Inconsistent data randomly received.

### Steps to reproduce
I can't make a project to reproduce the issue because you won't have the IOT Device.

### Stack trace and console log

No logs because the data sent on notification is not logged by this library.
I'm receiving a byte array with a length of 20.
Would it be possible to log this buffer in the iOS/Android logs?

I'm willing to fork the project, and do it myself, if I have instructions on how to integrate it in my app.
Thanks in advance.

marcosinigaglia, 2019-02-21:
Hi @omatrot , I don't understand, the advertising data is general, you have to check the device documentation.

omatrot, 2019-02-21:
@marcosinigaglia What I'm saying is that the advertising data structure is correct, but randomly, not the content. Furthermore, when connected to a PC the data received is always consistent with what is described in the device documentation. So there is a glitch somewhere in between...

closed 2019-06-18 by marcosinigaglia